FYSETC/SD-WIFI-PRO #6: SD WIFI PRO card doesn't start in UHS-II slot

readahead1, 2024-05-09:
I assume that the 2nd row pins of the SD WIFI PRO are interfering with the 2nd row of the UHS-II slot.
Potential workaround to set the 2nd row pins to floating, pull-up or pull-down and avoid any data traffic on those pins. (e.g. UART)  
Could you please provide your view on this issue? The issue occurs even without any firmware on the ESP32 chip.

Steps to reproduce:
1. idf.py erase-flash
3. insert card into UHS-II slot of a laptop (like Razer Blade 16)
3. Nothing happens

eliwong, 2024-08-05:
Please do not plug SD-WIFI-PRO into other UHS-II sockets, this is dangerous and may damage your device.
Because the pin definition of SD-WIFI-PRO is different from UHS-II.

readahead1, 2024-08-05:
> Please do not plug SD-WIFI-PRO into other UHS-II sockets, this is dangerous and may damage your device.
> 
> Because the pin definition of SD-WIFI-PRO is different from UHS-II.
> 
> 

Thank you for the answer! To reduce that risk of using the card in uhs-II slot, do you have a working proposal how to set the pins of the 2nd row floating?

eliwong, 2024-08-06:
Thanks for your research!
When we designed it, we didn't consider the second row of pins compatible with UHS-II at all. We just used the free pins in the second row to lead out more ESP IO for other possible uses.
For some devices using USH-II, if the IO of ESP32 provides unpredictable voltage levels, we are not sure what the device will do, so this is a dangerous operation.
Of course, for professionals, this may not be a problem.

readahead1, 2024-08-06:
Thank you! So it might or might not worked as I described. It wasn't consciously designed like that.

tiredboffin, 2024-08-06:
> > Please do not plug SD-WIFI-PRO into other UHS-II sockets, this is dangerous and may damage your device.
> > Because the pin definition of SD-WIFI-PRO is different from UHS-II.
> 
> Thank you for the answer!
> 
> I think I have got now how does it work: when inserting the card into standard UHS-II socket then it's EN and IO26 (IN1/IN2) pins are connected to ground pins of the socket:
> 
>     * having EN low, ESP32 will not start at all and so all other ESP32 pins exposed in the 2nd row of the socket stay floating
> 
>     * having IO26 low, the sd-chip will not be connected to the socket but to the sd-chip. The pins in the first row of the socket are not connected and stay floating as well. (Except the supply voltage pin)
> 
> 
> Because IO26 and EN are hardwired to the ground pin of a standard UHS-II socket, this behavior cannot be influenced from software...
> 
> Is this understanding correct?
> 
> I understand that it is better to stay away from standard UHS-II slots but I don't see yet how it will damage the host. The mechanics which I have described above seems to be quite safe for me...

To complete the picture:  on my card at minimum four fingers 11,12,13,14 (or 10,11,12,13 in standard UHS-II notation) has to be detached from EPS32 pins -- for example with help of a Scotch tape patch -- to make SD WiFi Pro work properly. So I guess fingers 11 and 14 (pins IO26 and EN)  are not enough to isolate as pins IO0 and IO2 connected to UHS-II diff signal line D0 (fingers 12,13) and hence are seen as LOW by EPS32 and this I think enforces flashing mode.  Scotch tape is not convenient though if the card is to be used with the development board so perhaps [SD Female - SD Male cable](https://www.amazon.com/Standard-Memory-Female-Extension-Extender/dp/B01M22RI5U/) would work better.

readahead1, 2024-08-06:
> > > Please do not plug SD-WIFI-PRO into other UHS-II sockets, this is dangerous and may damage your device.
> 
> > > Because the pin definition of SD-WIFI-PRO is different from UHS-II.
> 
> > 
> 
> > Thank you for the answer!
> 
> > 
> 
> > I think I have got now how does it work: when inserting the card into standard UHS-II socket then it's EN and IO26 (IN1/IN2) pins are connected to ground pins of the socket:
> 
> > 
> 
> >     * having EN low, ESP32 will not start at all and so all other ESP32 pins exposed in the 2nd row of the socket stay floating
> 
> > 
> 
> >     * having IO26 low, the sd-chip will not be connected to the socket but to the sd-chip. The pins in the first row of the socket are not connected and stay floating as well. (Except the supply voltage pin)
> 
> > 
> 
> > 
> 
> > Because IO26 and EN are hardwired to the ground pin of a standard UHS-II socket, this behavior cannot be influenced from software...
> 
> > 
> 
> > Is this understanding correct?
> 
> > 
> 
> > I understand that it is better to stay away from standard UHS-II slots but I don't see yet how it will damage the host. The mechanics which I have described above seems to be quite safe for me...
> 
> 
> 
> To complete the picture:  on my card at minimum four fingers 11,12,13,14 (or 10,11,12,13 in standard UHS-II notation) has to be detached from EPS32 pins -- for example with help of a Scotch tape patch -- to make SD WiFi Pro work properly. So I guess the fingers 11 and 14 (pins IO26 and EN)  are not enough to isolate as pins IO0 and IO2 are connected to UHS-II diff signal line D0 (fingers 12,13) and hence are seen as LOW by EPS32 and this I think enforces flashing mode.  Scotch tape is not convenient though if the card is to be used with the development board so perhaps [SD Female - SD Male cable](https://www.amazon.com/Standard-Memory-Female-Extension-Extender/dp/B01M22RI5U/) would work better.

I assume if secure boot is enabled, then IO02 and IO00 are ignored:

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/secure-boot-v1.html
I haven't tested it yet though.