FYSETC/SD-WIFI-PRO #7: Switching the multiplexer triggers re-mount of the sd-chip on the connected PC

readahead1, 2024-05-09:
The ESP32 chip runs a firmware which is switching the multiplex between the SD-slot and ESP32 chip. After the multiplexer is switched to ESP32 for more than 400 ms and then switched back to the SD-slot, the PC which is connected to Uploader & card reader board via USB remounts the chip. The same doesn't happen if the SD WIFI PRO is inserted into a plain UHS-I sd card reader from another vendor.

Steps to reproduce unexpected behavior:
1. flash a sample firmware to SD WIFI PRO which switches the sd chip via IO26 to the ESP32 and mounts and unmounts it periodically
2. Insert SD WIFI PRO into the Uploader & Card Reader Board and connect it via USB to a PC
3. Every time the sd-chip is return to the Uploaded & Card Reader board, the PC re-mounts the sd-chip

Steps to reproduce expected behavior:
1. flash a sample firmware to SD WIFI PRO which switches the sd chip via IO26 to the ESP32 and mounts and unmounts it periodically (same)
2. Insert the SD card into a plain UHS-I reader from another vendor
3. No re-mounts of the sd-chip on the PC is triggered while the PC is in idle

Already tries:
I realized that there are no 10k pull-up resistors on the GL823k chip if the multiplexer is in NO/ESP32. Added these 10 k pull-up resistors via the headers of the Uploader & Card Reader Board but it did not help.

Could you please advise what resistor/ capacity should I add to the headers of the Uploader & Card Reader Board in order to avoid this problem?

eliwong, 2024-08-06:
You mean

When you use the Uploader & card reader board to switch the storage chip from ESP32 to GL823K, the PC will remount the device. But it won't when using other card readers. Is that right?

I think this may have something to do with the way GL823K works. Some card readers will display the drive letter when there is no SD card, while others will not.

As for the 10K resistor you mentioned, GL823K does not have this resistor. Am I missing something?
The GL823 specification may help you:
[Genesys-GL823K.pdf](https://github.com/user-attachments/files/16507901/Genesys-GL823K.pdf)

readahead1, 2024-08-06:
According to my recent test, most of the card readers (other than gl823k) initiate a power cycle as recovery if they the sd chip is not responding because the ESP32 is using that. It is probably GL823k specific. I tried to add 10k pull-up resistors but it didn't help. I have also read on sdcard.org that pull-up resistors should be part of the host and not part of the card. So the current behavior seems to be optimal.

tiredboffin, 2024-08-07:
> According to my recent test, most of the card readers (other than gl823k) initiate a power cycle as recovery if they the sd chip is not responding because the ESP32 is using that. This is seamless for the host but ESP32 needs to deal with the unexpected restart. Overall it is an optimal behavior.
> 

1) It perhaps is not that unique in this regard - in the following two readers SD WifI Pro also does not reboot when ESP32 takes control over SD chip even for minutes-- time enough to upload/download about 1G file over wifi .  

[image]

[image]

The card stays mounted on PC side and if the host tries to read  from the card  (I used dd if=/dev/sdc1 in my test)  at the time of the transfer i.e. when the SD is under ESP32 control, the kernel reports the following I/O errors: 

[image]
 
2) The reader board when used as SD card reader though reports "capacity change" to PC in about a second after EPS32 takes the control.

[image]

and udisksd tries to re-read partition table, fails and cleans up the mount point.

It reports "capacity change" again when EPS32 relinquishes the control and and udisksd mounts the card as if it was physically re-inserted:

[image]

3) In the following generic noname reader SD Wifi Pro card does reboot as you described:

[image]

The reader first reports "capacity change to 0" (similar to the board case (2) above) followed by immediate "capacity change from 0"
[image]

This is the least desired behaviour for my use case.

readahead1, 2024-08-07:
> > According to my recent test, most of the card readers (other than gl823k) initiate a power cycle as recovery if they the sd chip is not responding because the ESP32 is using that. This is seamless for the host but ESP32 needs to deal with the unexpected restart. Overall it is an optimal behavior.
> 
> 1. It perhaps is not that unique in this regard - in the following two readers SD WifI Pro also does not reboot when ESP32 takes control over SD chip even for minutes-- time enough to upload/download about 1G file over wifi .
> 
> ![image](https://private-user-images.githubusercontent.com/17812670/355789556-9474c703-be57-4e56-bd0f-c5b46014756c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNjAzMTQsIm5iZiI6MTcyMzA2MDAxNCwicGF0aCI6Ii8xNzgxMjY3MC8zNTU3ODk1NTYtOTQ3NGM3MDMtYmU1Ny00ZTU2LWJkMGYtYzViNDYwMTQ3NTZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDE5NDY1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhMTgwMDQ5NmEzM2QwZTQxNmZmYjQ5ODY1MTNhMjg5OTZmNjY5YTRkNDVlMzNmOWJiMWQ1ZjNkMTllZDJmMmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4I5zmsm2NLvDdkqPkN3HOMzpGGlMxIRzFMUedk5Gr9o)
> 
> ![image](https://private-user-images.githubusercontent.com/17812670/355806304-919df532-d274-45f0-93f1-52e312e4ebd6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNjAzMTQsIm5iZiI6MTcyMzA2MDAxNCwicGF0aCI6Ii8xNzgxMjY3MC8zNTU4MDYzMDQtOTE5ZGY1MzItZDI3NC00NWYwLTkzZjEtNTJlMzEyZTRlYmQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDE5NDY1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZiNWQxZjJiMDE3M2QwMmY5ZTAzZTZkMGRjN2JhM2E4MDMyMTE5NDMwNDU1MGVjMTBkYjI5MzQ2OWRkMmRhMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.c_9jRc2O8cp9yr_ILawVhQTj3gO8zwAFoLIJTMycWiY)
> 
> The card stays mounted on PC side and if the host tries to read from the card (I used dd if=/dev/sdc1 in my test) at the time of the transfer i.e. when the SD is under ESP32 control, the kernel reports the following I/O errors:
> 
> ![image](https://private-user-images.githubusercontent.com/17812670/355793776-545eab0f-f40d-4211-8686-ed2846c4e95d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNjAzMTQsIm5iZiI6MTcyMzA2MDAxNCwicGF0aCI6Ii8xNzgxMjY3MC8zNTU3OTM3NzYtNTQ1ZWFiMGYtZjQwZC00MjExLTg2ODYtZWQyODQ2YzRlOTVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDE5NDY1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiMmI3YTU5NDk5NTg5Nzg0ZTNjNGJiZjgwYzAzNzFlMzZmZmMyMWI4ZjUzNjFhMjAzOTdiZDA4ZDBlZGQ5MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2La7s2AyHvSgzul2NkWd58vWzggDmPg1_RYvIUMpBik)
> 
> 2. The reader board when used as SD card reader though reports "capacity change" to PC in about a second after ESP32 takes the control.
> 
> ![image](https://private-user-images.githubusercontent.com/17812670/355800110-e3cb694a-b799-44cb-82cc-47344ce1cb8f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNjAzMTQsIm5iZiI6MTcyMzA2MDAxNCwicGF0aCI6Ii8xNzgxMjY3MC8zNTU4MDAxMTAtZTNjYjY5NGEtYjc5OS00NGNiLTgyY2MtNDczNDRjZTFjYjhmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDE5NDY1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdmNzc1Nzk5YjdiYTg3NGIxMmQyODczMzNlMmRjMmZmNTU1N2ZiOTg1Mjk1N2Y0MDc1MzYzOTYwN2ZmMGY5ZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Z0znF6veDdjkGo6TlfCBc1b4A43PGToT2UHpHZg8PN0)
> 
> and udisksd tries to re-read partition table, fails and cleans up the mount point.
> 
> It reports "capacity change" again when ESP32 relinquishes the control and and udisksd mounts the card as if it was physically re-inserted:
> 
> ![image](https://private-user-images.githubusercontent.com/17812670/355805059-ef5b2e60-1220-4455-b997-2bf1988cbdc2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNjAzMTQsIm5iZiI6MTcyMzA2MDAxNCwicGF0aCI6Ii8xNzgxMjY3MC8zNTU4MDUwNTktZWY1YjJlNjAtMTIyMC00NDU1LWI5OTctMmJmMTk4OGNiZGMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDE5NDY1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkMjI1ZTQ1MWZlNTJkN2UyNTljNzE1MjM2NjdlOWVkYTZiYTc0MWQ3OTlmNWQwMjU2YWUzMzdhZGUxMTg0MDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.y5kAIiEZl-w3Q6KSWwTFFQABEsG601NhCc3L6Sw54v0)
> 
> 3. In the following generic noname reader SD Wifi Pro card does reboot as you described:
> 
[... cut ...] 3ZDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HItzX0RLOdA2kk-e01_FL0euQo4p4BTATC1ElAU4XLA)
> 
> This is the least desired behaviour for my use case.

This is a great overview thank you!

In the 1st case when kernel reports the I/O errors and neither reboot nor re-mount happens:
does the communication between card and PC auto-recovers after the ESP32 is not using the sd-chip anymore? In my case when I/O error occurs, then I need to manually re-insert the card...

tiredboffin, 2024-08-08:
> This is a great overview thank you!
>
> In the 1st case when kernel reports the I/O errors and neither reboot nor re-mount happens: does the communication between card and PC auto-recovers after the [...]

I modified my ESP32 code to log the reboots properly and now from the logs I can see that in case (1) above esp32 does reboot after all. My initial analysis for the case (1) was wrong as dd uses read cache by default and in my quick experiment it did not trigger a collision as it was reading from the cache. Sorry for the confusion. I have disabled the cache and can confirm that if there is a collision then both readers in (1) reboot the card _silently_. On the host side these collisions/reboots may get completely unnoticed (as my esp32 code relinquishes the control first thing in main()) or occasionally I observe I/O errors in the host logs. If the host does not use the card or it hits the cache at the time SD is used by EPS32  then SD card stays mounted, ESP32 is not rebooted, I see no errors in the host logs -- no collision no problem.

Case (3) is still a bit different as the reader _unconditionally_ reboots ESP32 about a second or two after ESP32 takes the control   -- when the card is not used by the host and is not even mounted on the host side. This reader is more problematic even when there are no true collisions.